markrossington/sidewinder-x2-marlin #4: Problems compiling firmware

eRaz3r, 2023-02-11:
When I try to compile the firmware, I get thsi error
`[Info] Running C:\Users\Raz3r/.platformio/penv/bin/pio run
Traceback (most recent call last):
  File "C:\sidewinder-x2-marlin\scripts\01-build.py", line 147, in <module>
    main()
  File "C:\sidewinder-x2-marlin\scripts\01-build.py", line 141, in main
    if not mb.build_marlin():
           ^^^^^^^^^^^^^^^^^
  File "C:\sidewinder-x2-marlin\scripts\01-build.py", line 122, in build_marlin
    self.run_process([self.pio_command, "run"])
  File "C:\sidewinder-x2-marlin\scripts\01-build.py", line 61, in run_process
    with subprocess.Popen(process_command, stdout=subprocess.PIPE, stderr=subprocess.STDOUT) as process:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Raz3r\AppData\Local\Programs\Python\Python311\Lib\subprocess.py", line 1024, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\Raz3r\AppData\Local\Programs\Python\Python311\Lib\subprocess.py", line 1493, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [WinError 2] Das System kann die angegebene Datei nicht finden`

I reinstall python and tried a few other things without success. Maybe you can help.

markrossington, 2023-02-11:
Hello, first off congrats on being the first to report an issue to this repo (other than me).

Hmmm ok - I'll look into this. I think perhaps you have an existing install of platformio on the path. A few questions:

* Did it install platformio or did you have it installed already?
* Do you have anything in `C:\Users\Raz3r\.platformio` folder?
* What are the results of typing `which pio` on the command prompt?

You can fix this by modifying line 26 of `01-build.py` to the path of wherever you have platformio already, or maybe just `pio` if it's already on the PATH. e.g.

```python
    pio_command = "pio"
```

eRaz3r, 2023-02-11:
Thank you very much for your fast reply.

>     * Did it install platformio or did you have it installed already?
Yes it did I think because..................
>     * Do you have anything in `C:\Users\Raz3r\.platformio` folder?
........there are the folders .cache and penv inside
>     * What are the results of typing `which pio` on the command prompt?
that "pio" isn't known.

> You can fix this by modifying line 26 of 01-build.py to the path of wherever you have platformio already, or maybe just pio if it's already on the PATH. e.g.
The folder bin/ is missing so is pio. Any Idea how to fix this?

Thank you very much

markrossington, 2023-02-11:
Delete the whole `.platformio` folder and start the script again. 

Try to save all of what the script  says and post it here if it fails again :-)

eRaz3r, 2023-02-11:
I got the same issue like before. Is there a possibility to install PlatformIO serperatly?

markrossington, 2023-02-11:
Yep, try this https://docs.platformio.org/core/installation/methods/installer-script.html

Still I don't understand how the code is getting there. It should check for the platform io binary and only continue if it exists. I'll need to look into this as that's definitely a bug.

eRaz3r, 2023-02-12:
I tried to use the script or point to pio in the script without success. Also the einviroment vaiable didn't make it. Pio in my case is located in 
> C:\Users\Raz3r\.platformio\penv\Scripts 
but pointing there in the script doesn't solve the problem.

markrossington, 2023-02-13:
Thanks for the PR. I should get to try it on a windows machine in a day or so. 

Please note the flash script isn't working properly yet so please do this manually for now. I hadn't expected people to find my repo already 😅

eRaz3r, 2023-02-13:
Allright. Good to know but I have to wait till my jumpers arrive. Actual I am on klipper and want to go back. The problem is, that the dfu mode only can entered with pronterface when you are on Marlin. When you are on clipper it is neccesary to insert the jumper. For my bad I first see your firmware after switching to klipper. With the enabled features in your release (f.e. filament change) I would have stayed on stock fw.

markrossington, 2023-02-13:
Should work now, try again :)

commit 605247fb (2023-02-13): fix: make pio_command cross platform

fixes #4, fixes #7
closed 2023-02-13 by markrossington