markrossington/sidewinder-x2-marlin #7: WinError 2

2mitts, 2023-02-13:
I think I'm getting through most of this but hanging up at a save point. Tried a few different directory locations to run from but no changes. Very inexperienced user here so may be something simple. Also, i made the mistake of not clicking "add python to path" when installing Python originally, and found it that was necessary. Recommend adding that to your instructions for us rookies. info below.  Thank you for putting this together. 

C:\Users\mwesa\OneDrive\Desktop\sidewinder-x2-marlin-main>python scripts/01-build.py
[Info] Setting working directory to C:\Users\mwesa\OneDrive\Desktop\sidewinder-x2-marlin-main
[Info] Creating directories:
        tmp
        build
        output
[Info] Checking for C:\Users\mwesa/.platformio/penv/bin/pio
[Info] Downloading https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py to tmp/get-platformio.py...
[Info] Running C:\Users\mwesa\AppData\Local\Programs\Python\Python311\python.exe tmp/get-platformio.py
Installer version: 1.1.2
Platform: Windows-10
Python version: 3.11.2 (tags/v3.11.2:878ead1, Feb  7 2023, 16:38:35) [MSC v.1934 64 bit (AMD64)]
Python path: C:\Users\mwesa\AppData\Local\Programs\Python\Python311\python.exe
Creating a virtual environment at C:\Users\mwesa\.platformio\penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio
Using cached platformio-6.1.6-py3-none-any.whl
Collecting bottle==0.12.*
Using cached bottle-0.12.23-py3-none-any.whl (90 kB)
Collecting click<9,>=8.0.4
Using cached click-8.1.3-py3-none-any.whl (96 kB)
Collecting colorama
Using cached colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Collecting marshmallow==3.*
Using cached marshmallow-3.19.0-py3-none-any.whl (49 kB)
Collecting pyelftools<1,>=0.27
Using cached pyelftools-0.29-py2.py3-none-any.whl (174 kB)
Collecting pyserial==3.5.*
Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting requests==2.*
Using cached requests-2.28.2-py3-none-any.whl (62 kB)
Collecting semantic-version==2.10.*
Using cached semantic_version-2.10.0-py2.py3-none-any.whl (15 kB)
Collecting tabulate==0.9.*
Using cached tabulate-0.9.0-py3-none-any.whl (35 kB)
Collecting aiofiles==22.1.*
Using cached aiofiles-22.1.0-py3-none-any.whl (14 kB)
Collecting ajsonrpc==1.*
Using cached ajsonrpc-1.2.0-py3-none-any.whl (22 kB)
Collecting starlette==0.23.*
Using cached starlette-0.23.1-py3-none-any.whl (64 kB)
Collecting uvicorn==0.20.*
Using cached uvicorn-0.20.0-py3-none-any.whl (56 kB)
Collecting wsproto==1.2.*
Using cached wsproto-1.2.0-py3-none-any.whl (24 kB)
Collecting packaging>=17.0
Using cached packaging-23.0-py3-none-any.whl (42 kB)
Collecting charset-normalizer<4,>=2
Using cached charset_normalizer-3.0.1-cp311-cp311-win_amd64.whl (96 kB)
Collecting idna<4,>=2.5
Using cached idna-3.4-py3-none-any.whl (61 kB)
Collecting urllib3<1.27,>=1.21.1
Using cached urllib3-1.26.14-py2.py3-none-any.whl (140 kB)
Collecting certifi>=2017.4.17
Using cached certifi-2022.12.7-py3-none-any.whl (155 kB)
Collecting anyio<5,>=3.4.0
Using cached anyio-3.6.2-py3-none-any.whl (80 kB)
Collecting h11>=0.8
Using cached h11-0.14.0-py3-none-any.whl (58 kB)
Collecting sniffio>=1.1
Using cached sniffio-1.3.0-py3-none-any.whl (10 kB)
Installing collected packages: pyserial, pyelftools, charset-normalizer, bottle, urllib3, tabulate, sniffio, semantic-version, packaging, idna, h11, colorama, certifi, ajsonrpc, aiofiles, wsproto, requests, marshmallow, click, anyio, uvicorn, starlette, platformio
Successfully installed aiofiles-22.1.0 ajsonrpc-1.2.0 anyio-3.6.2 bottle-0.12.23 certifi-2022.12.7 charset-normalizer-3.0.1 click-8.1.3 colorama-0.4.6 h11-0.14.0 idna-3.4 marshmallow-3.19.0 packaging-23.0 platformio-6.1.6 pyelftools-0.29 pyserial-3.5 requests-2.28.2 semantic-version-2.10.0 sniffio-1.3.0 starlette-0.23.1 tabulate-0.9.0 urllib3-1.26.14 uvicorn-0.20.0 wsproto-1.2.0
PlatformIO Core has been successfully installed into an isolated environment `C:\Users\mwesa\.platformio\penv`!
The full path to `platformio.exe` is `C:\Users\mwesa\.platformio\penv\Scripts\platformio.exe`
If you need an access to `platformio.exe` from other applications, please install Shell Commands
(add PlatformIO Core binary directory `C:\Users\mwesa\.platformio\penv\Scripts` to the system environment PATH variable):
See https://docs.platformio.org/page/installation.html#install-shell-commands
[Info] Cleaning up:
        tmp/get-platformio.py
[Info] Downloading https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip to tmp/marlin_bugfix-2.1.x.zip...
[Info] Cleaning up build directory
[Info] Extracting tmp/marlin_bugfix-2.1.x.zip to build folder...
[Info] Cleaning up:
        tmp/marlin_bugfix-2.1.x.zip
[Info] Copying config/Configuration.h to build\Marlin-bugfix-2.1.x\Marlin
[Info] Copying config/Configuration_adv.h to build\Marlin-bugfix-2.1.x\Marlin
[Info] Making build\Marlin-bugfix-2.1.x\platformio.ini file build for Ruby board
[Info] Building Marlin found in build\Marlin-bugfix-2.1.x
[Info] Running C:\Users\mwesa/.platformio/penv/bin/pio run
Traceback (most recent call last):
  File "C:\Users\mwesa\OneDrive\Desktop\sidewinder-x2-marlin-main\scripts\01-build.py", line 147, in <module>
    main()
  File "C:\Users\mwesa\OneDrive\Desktop\sidewinder-x2-marlin-main\scripts\01-build.py", line 141, in main
    if not mb.build_marlin():
           ^^^^^^^^^^^^^^^^^
  File "C:\Users\mwesa\OneDrive\Desktop\sidewinder-x2-marlin-main\scripts\01-build.py", line 122, in build_marlin
    self.run_process([self.pio_command, "run"])
  File "C:\Users\mwesa\OneDrive\Desktop\sidewinder-x2-marlin-main\scripts\01-build.py", line 61, in run_process
    with subprocess.Popen(process_command, stdout=subprocess.PIPE, stderr=subprocess.STDOUT) as process:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\mwesa\AppData\Local\Programs\Python\Python311\Lib\subprocess.py", line 1024, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\mwesa\AppData\Local\Programs\Python\Python311\Lib\subprocess.py", line 1493, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [WinError 2] The system cannot find the file specified

markrossington, 2023-02-13:
Currently doesn't work on windows. Look at pull request #5 for fix. I should be able to test and fix on a windows machine soon. Sorry about this.

markrossington, 2023-02-13:
Also please note the flash script isn't yet working properly. I hadn't suspected people would find this repo already 😅

2mitts, 2023-02-13:
Apologies for not seeing that, I'll take a look. Found your project through google yesterday, apparently they're indexing pretty quick. Appreciate your efforts.

markrossington, 2023-02-13:
Should work now, try again :)

closed 2023-02-13 by markrossington

eRaz3r, 2023-02-13:
Like Mark says there was a problem with the path to pio. After he merged the PR to the main repository, the build script should word as espected.

commit 605247fb (2023-02-13): fix: make pio_command cross platform

fixes #4, fixes #7