markrossington/sidewinder-x2-marlin #10: Add what to do if not going into dfu-mode

markrossington, 2023-02-14:
Allegedly some sidewinder x2 need a jumper and others can just reboot into dfu mode. 

> If this command fails, then you need to put the machine in DFU mode using the jumper pin on the board.

To do this, you need to remove the bottom cover of the printer in order to gain physical access to the Artillery Ruby board. Then, you need to install a jumper to connect the BOOT and the +3.3V pin. This connection will put the Ruby board in DFU mode and will allow you to flash the printer.

From https://3dprintbeginner.com/how-to-install-klipper-on-sidewinder-x2/

eRaz3r, 2023-02-17:
Maybe add a hint, that if you want get back from klipper (like me), it is also neccesary to use a jumper if your board first don't need any. After you install Klipper f.e., the DFU mode is not accesible via Pronterface (M997).

markrossington, 2023-02-17:
Oh that's a very good idea. Does klipper not have any equivalent then?

closed 2023-02-17 by markrossington

eRaz3r, 2023-02-18:
Maybe it has but I didnt find anything that works.

markrossington, 2023-02-18:
The Klipper code seems to attempt to enter dfu mode a different way. 

See here https://github.com/Klipper3d/klipper/blob/master/scripts/flash_usb.py

I will look at the platformio code too. 

Might even use/repurpose their flash scripts if they can force into dfu mode for you.

commit def104c3 (2023-02-17): docs: add a manual flashing guide

fixes #10, fixes