markrossington/sidewinder-x2-marlin #9: Get flash script to base level functionality

markrossington, 2023-02-14:
First use `dfu-util -l` to see if the device is in dfu mode. Or poll this to wait after an `m997` command. Could also add the vendor ID but would need to check this is always the same. 

Second make the download command reset the device once complete with the `-R` switch. 

Close the serial port once dfu-util finds the printer and reopen it once download is complete. 

Need to add some conditional steps and error checking at each stage.

markrossington, 2023-02-17:
Good shout. Could actually use stmcube on the command line too 😂

markrossington, 2023-02-17:
See the new manual flashing guide just added. I add a link to artillery guide which mentions STM32CubeProgrammer https://github.com/markrossington/sidewinder-x2-marlin/blob/main/flashing.md

eRaz3r, 2023-02-18:
Thats it 😁.

markrossington, 2023-02-18:
`~/.platformio/penv/bin/pio run -t upload -e Artillery_Ruby`

The upload command uses something called openOCD which will be installed when building the firmware anyway. This will remove the dependency on dfu-util. If it works. It might also do a better job of putting the Ruby board in reprogram mode.

The openocd command that pio calls is this... which atm doesn't mean very much to me.

`openocd -d2 -f scripts/interface/stlink.cfg -c "transport select hla_swd" -f scripts/target/stm32f4x.cfg -c "reset_config none" -c "program {.pio/build/Artillery_Ruby/firmware.elf}  verify reset; shutdown;`

markrossington, 2023-03-06:
With using platform's version of dfu-util (#17) and some minor changes 9dc23b33fa75e422af6f9244fba5d30435f2e43f the script is looking a little better. I'd say it's at base level, NOT tested, level now.

markrossington, 2023-03-06:
@eRaz3r If you have your own config, try looking at `marlin_configuration_h` and `marlin_configuration_adv_h` configuration options if you have your own marlin config

closed 2023-03-06 by markrossington