markrossington/sidewinder-x2-marlin #13: No heating bed or extruder

2mitts, 2023-02-19:
I was able to successfully compile your firmware, and get it installed via the instructions you provided. Everything appears to be working correctly, but i have not been able to get a file to start printing, the bed/extruder heating settings for the file are shown on the screen, but the heating process is never started/completed.  I am able to set temps manually through the printer UI as well as through pronterface. I have attached a screen shot in hopes it may be helpful. Ty for the time and energy you are putting into this project. 
[image]

labotecno, 2023-02-19:
this version  working with only tft bbt  , try to take off host action command on  config adv

2mitts, 2023-02-19:
I was just coming back to make that update, i'm sure it's a TFT issue . . . I will try your fix, appreciate the information.

markrossington, 2023-02-19:
What's TFT bbt? Maybe BigTreeTech TFT firmware?

Are you trying to print with SD card or via serial codes? (e.g. octoprint?)

If there's an incompatibility with display firmware I'd like to try and fix it for you folks. I think `HOST_ACTION_COMMANDS` is useful for octoprint users though.

2mitts, 2023-02-19:
I appreciate your willingness to help, ty. I have the GD32F305 TFT. I am printing with a USB drive/stick, no ocotoprint or remote setup. I found two other firmware builds that are supposed to work without touching the TFT side of things, but i'm having the same issue with one of them, and the other works (Mr. Magounet) but has extremely slow probe settings that aren't going to work with the fast paced work flow here.

markrossington, 2023-02-19:
Hard to compare to his as his is a much older version of marlin

https://github.com/MrMagounet/Firmware_Flash-Artillery_X2_GeniusPro/tree/main/marlin/Marlin

I'll try to see what's different.

labotecno, 2023-02-19:
> What's TFT bbt? Maybe BigTreeTech TFT firmware?
> 
> Are you trying to print with SD card or via serial G-codes? (e.g. octoprint?)
> 
> If there's an incompatibility with display firmware I'd like to try and fix it for you folks. I think `HOST_ACTION_COMMANDS` is useful for octoprint users though.

Yes bigtreetech tft firmware,  I m getting problem when I try printing with sd card and USB without modificate the tft firmware stay stuck to getting hotend and heat bed at temperature.  I m getting working take off some options . By the way your system to built the firmware  is really good it will be working for other mainboard I think  . Ps my printer have New tft . Thank you for your precious work!

2mitts, 2023-02-20:
Labotecno; have you had luck with freakyDude's firmware that was built specifically for our tft screen versions? I am not having luck with his either.

markrossington, 2023-02-20:
I am unable to determine what in his firmware vs mine means yours works. Perhaps it's a bug in the newer version of Marlin.

Do you have the gcode you're trying to print?

markrossington, 2023-02-20:
Tomorrow I'll pull a fresh version and build it. I can then test a print to see if there's a problem in the configuration. Maybe I missed something/didn't commit something.

2mitts, 2023-02-20:
Went through freakDude's latest, and while v2.1.2.0  doesn't appear to be working for me, however v2.1.0.0 seems to be behaving as expected.

markrossington, 2023-02-20:
Hmm input shaping was added in 2.1.2 maybe it's related?

labotecno, 2023-02-20:
Firmware from magounet , freakydude dont working tooo . I get problem to printing tooo . After I will upload my firmware to show you its working without a some option .

markrossington, 2023-02-27:
This seems to affect you with other builds of marlin than mine and so I will close it for now until there's any new information.

markrossington, 2023-03-02:
Cool just compared. Not massively different, mostly advanced commands off and turning some software PWM features on for the fans.

I note you don't have a second Z axis stepper configured. Any reason? 
Also note your Z probe offset is much further from the nozzle. Did you move it?

labotecno, 2023-03-03:
I note you don't have a second Z axis stepper configured. Any reason?
-i let the system with belt synchronization genuine , you need to take of the belt for G34 

 your Z probe offset is much further from the nozzle. Did you move it?
-yes because its with bltouch normal . 40 for y axis 
 this firmware working for last version of lcd gd 

try it and tell me is all its ok