markrossington/sidewinder-x2-marlin #20: Cannot compile the firmware

Firnthon, 2023-06-15:
I'm trying to compile the firmware on Pop!_OS 22.04 LTS, Python 3.10.6.

Output: AttributeError: module 'settings' has no attribute 'local_pio_script_path'.

markrossington, 2023-06-16:
Hmmm interesting. Silly question perhaps: do you have the `settings.py` file?

Firnthon, 2023-06-16:
Yes, in the scripts folder.

Firnthon, 2023-06-16:
Full log:

`jakub@pop-os:~/Pobrane/sidewinder-x2-marlin-main$ sudo python3 scripts/marlin_build.py
[sudo] hasło użytkownika jakub: 
[Info] Cleaning up tmp/custom_config directory
[Info] Setting working directory to /home/jakub/Pobrane/sidewinder-x2-marlin-main
[Info] Creating directories:
	tmp
	build
	output
[Info] Checking for /root/.platformio/penv/bin/pio
[Info] /root/.platformio/penv/bin/pio not found
Traceback (most recent call last):
  File "/home/jakub/Pobrane/sidewinder-x2-marlin-main/scripts/marlin_build.py", line 111, in <module>
    main()
  File "/home/jakub/Pobrane/sidewinder-x2-marlin-main/scripts/marlin_build.py", line 101, in main
    Common.install_platformio()
  File "/home/jakub/Pobrane/sidewinder-x2-marlin-main/scripts/common.py", line 104, in install_platformio
    Common.download_file(settings.pio_download_url, settings.local_pio_script_path)
AttributeError: module 'settings' has no attribute 'local_pio_script_path'`

dgalli1, 2023-06-25:
I believe https://github.com/markrossington/sidewinder-x2-marlin/commit/d32136243abb36082b14d6f89caaa59fbe3ff612 missed this setting.

Writting this into it fixed it (path is taken from [common.py](https://github.com/markrossington/sidewinder-x2-marlin/blob/48814ffceeabbf57e561d9cc12247478de5adc0e/scripts/common.py#L18))
local_pio_script_path = "tmp/get-platformio.py"

markrossington, 2023-07-05:
Thanks I'll work this in to my next update.

saulmayo, 2023-07-05:
Can you explain how you have solved it? I have the same problem.

Firnthon, 2023-07-05:
As adviced in https://github.com/markrossington/sidewinder-x2-marlin/issues/20#issuecomment-1606255068, copy paste local_pio_script_path = "tmp/get-platformio.py" to scripts/settings.py (I've added to line 34).

saulmayo, 2023-07-05:
> Como se aconseja en [ el n. ° 20 (comentario) ](https://github.com/markrossington/sidewinder-x2-marlin/issues/20#issuecomment-1606255068) , copiar y pegar `local_pio_script_path = "tmp/get-platformio.py"`a scripts/settings.py (he agregado a la línea 34).

yes now! thank you very much!

artlomako, 2023-07-15:
@markrossington hello, I tries to create a Pull Request with fix, but I it is not possible - It seems like I do not have permissions. Could you please allow branches/PRs creation for other users?

markrossington, 2023-07-15:
That's unusual, I haven't changed any settings and have accepted other people's pull requests, e.g. #5

markrossington, 2023-07-16:
Sorted now, sorry for the delay.

PR #22 (2023-07-16): fix(scripts): add pio script path back in
fixes #20
closed 2023-07-16 by markrossington