markrossington/sidewinder-x2-marlin #21: #error "Don't build with import-2.1.x configurations!"

amibumping, 2023-07-05:
Hi, I tried compiling with default configurations and it worked perfectly, but if I try to use the latest configurations on Marlin https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Artillery/Sidewinder%20X2 I recived the error:

```
[pio] buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:23:2: error: #error "Don't build with import-2.1.x configurations!"
	[pio] 23 | #error "Don't build with import-2.1.x configurations!"
	[pio] |  ^~~~~
	[pio] buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:24:2: error: #error "Use the 'bugfix...' or 'release...' configurations matching your Marlin version."
	[pio] 24 | #error "Use the 'bugfix...' or 'release...' configurations matching your Marlin version."
	[pio] |  ^~~~~
	[pio] Error: Failed to parse Marlin features. See previous error messages.
	[pio] ========================== [FAILED] Took 0.65 seconds ==========================
	[pio] Environment     Status    Duration
	[pio] --------------  --------  ------------
	[pio] Artillery_Ruby  FAILED    00:00:00.652
	[pio] ==================== 1 failed, 0 succeeded in 00:00:00.652 ====================
```
Thank you!

amibumping, 2023-07-05:
Sorry I forgot that I don't read before I ask... I see that yours it's a modified version of the original one.
Just one question, do you plan to update the configs any time?

Thank you for the work.

closed 2023-07-05 by amibumping

markrossington, 2023-07-05:
Sorry yeah I do plan to but I have been busy. Thanks for the reminder :-)

markrossington, 2023-07-16:
See #23 

It at least compiles with latest bugfix-2.1.x branch as of today. I haven't tried it on my printer yet and so try with caution. I will merge when tested.