markrossington/sidewinder-x2-marlin #26: Compile to .hex for STM32Programmer

aweussom, 2023-08-15:
My SWX2 turns up in Device Manager as STM32Bootloader after M997.

I need the firmware as a .hex file to program using STM32 Programmer - how can I generate that?

aweussom, 2023-08-15:
I caved in an put a jumper on boot/3.3v and flashed .bin using dfu-util.exe.

closed 2023-08-15 by aweussom

markrossington, 2023-08-15:
You could have used dfu-util for the first device I believe.

aweussom, 2023-08-16:
> You could have used dfu-util for the first device I believe. (In stm32bootloader mode)

Sorry, no. 
dfu-util -l 

did not list the STM when it was in STM32Bootloader mode.
Needed to jumper it to "boot" before it turned up.

aweussom, 2023-08-16:
Other than that, excellent work!

The flash script does not work on Windows, but you probably knew that :-)