markrossington/sidewinder-x2-marlin #34: distutils removed in Python 3.12

nharrer, 2024-01-28:
The deprecated distutils has been removed in Python 3.12 (see https://docs.python.org/3.10/library/distutils.html). So the build script throws:

```
Traceback (most recent call last):
  File "C:\devtools\sidewinder-x2-marlin\scripts\marlin_build.py", line 11, in <module>
    from common import Common
  File "C:\devtools\sidewinder-x2-marlin\scripts\common.py", line 7, in <module>
    from distutils.spawn import find_executable
ModuleNotFoundError: No module named 'distutils'
```

I think a workaround is to install a recent setuptools: `pip install setuptools`. 

Or, what I did was to replace in `common.py`:
```
find_executable(command)
```
with 
```
shutil.which(command)
```

BTW: Awesome project!! It is such a big help for Sidewinder owners who are not very familiar with compiling Marlin and all the configuration settings. THANK YOU!!

Zogar89, 2024-02-23:
@markrossington @nharrer The first solution works for my.

On cmd (windows console) pip install setuptools

markrossington, 2024-02-23:
Apologies both I missed the notification on this. 

Thanks for the detailed issue. Yes I'll use an alternative standard library method for this function. The hope is that this script should work without the user needing any dependencies.

I can hopefully look at this next week 👍🏽

PR #35 (2024-03-24): fix: switch distutils.find_executable with shutil.which
Fixes #34
closed 2024-03-24 by markrossington